buttplugio/stpihkal #13: Add Onyx 2 Info

qdot, 2017-12-31:
Service UUID: f60402a6-0294-4bdb-9f20-6758133f7090

Charactertistic UUIDs:

Data: 02962ac9-e86f-4094-989d-231d69995fc2
Sensor Notification: d44d0393-0731-43b3-a373-8fc70b1f3323
Command/Firmware: c7b7a04b-2cc4-40ff-8b10-5d531d1161db

Data format: Takes 2 byte commands, first byte usually 0x32 or 0x63, possibly denoting position? Second byte varies.

Sensor Notification Format: First byte range 0x0-0x63, denotes touch position on pad if being touched, second byte: 0x0: no touch, 0x1: touch, 0x2: ?

blackspherefollower, 2018-02-05:
Looking at the sniff log, the sensor only reports 0,25,50,75,100 (havn't actually seen 75 yet, but I assume it's just the original 0-4 scaled up to 0-100.

blackspherefollower, 2018-05-22:
It's the launch 1.2 protocol (smacks head hard on desk)
Cmd: 0x00 to initialize
Control by sending 0xXX,0xYY to the data Char
   XX = position 0-100 (0x00-0x63) bigger = nearer
   YY = speed 0-100 (0x00-0x63) bigger = faster

I'll measure the speed later to see if this also matches the launch.

commit 4cd51c2c (2018-05-25): Add Kiiroo 2 documentation

Just add UUIDs and link protocol to fleshlight documentation

Fixes #13
closed 2018-05-25 by qdot