buttplugio/stpihkal #18: Lovense Toy Model Type letters needed

Darkhub, 2018-02-16:
I need letters for lovense model types, can anybody help?
The link https://github.com/metafetish/stpihkal/blob/master/stpihkal/hardware/models.md isn't working.

Thanks in advance. :)

qdot, 2018-02-16:
Oops. I thought I moved the letter identifer list over. >.>

Edge/Ambi/Domi/Osci in this list haven't been tested.

```
+------+-------------------------+---------+-------------------------------------+
| Name | Form Factor             | Model   | Capabilities                        |
+======+=========================+=========+=====================================+
| Nora | Rabbit Vibe             | A or C  | Vibration, Rotation, Accelerometer  |
+------+-------------------------+---------+-------------------------------------+
| Max  | Onahole                 | B       | Vibration, Inflation, Accelerometer |
+------+-------------------------+---------+-------------------------------------+
| Lush | Vaginal Insertable Vibe | S       | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
| Hush | Buttplug                | Z       | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
| Edge | Prostate Stimulator     | P | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
| Ambi | Bullet Vibe             | L | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
| Domi | Magic Wand Style Vibe   | W | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
| Osci | G-Spot Vibrator         | O | Vibration                           |
+------+-------------------------+---------+-------------------------------------+
```

Will close this once I actually get the table added to the documentation too.

Darkhub, 2018-02-16:
Thanks, I notice a strange thing happening, even if I reply at DeviceType request with the string "B:11:000000000001;" my device is reported in lovense androis app as hush, not max.

Everything should be correct, according to documentation I've declared:
#define SERVICE_UUID "0000fff0-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_UUID_RX "0000fff1-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_UUID_TX "0000fff2-0000-1000-8000-00805f9b34fb"

Any idea on how to fix? My suspect is on firmware version, maybe max has different firmware revision?

qdot, 2018-05-19:
@Darkhub I was just fixing this documentation bug and realized I never answered your question. Did you get this fixed? Are you trying to emulate lovense toys?

closed 2018-05-19 by qdot

Darkhub, 2018-05-21:
@qdot Yes, I am trying to emulate lovense toys on cheap hardware, this will also permit to use any other vibrating toy with lovense functionalities.
But the app detects the toy always as lush, no way to make it work as a max, I'd like to investigate the cause but I need some help from a max user that can provide me a bluetooth hci log of the data traffic between smartphone and toy...

qdot, 2018-05-21:
@darkhub I can definitely help on that (I've got all the Lovense toys but the Ambi), and @mftrips, the developer of metafetish/buttplug-android also made a spoofing app for Android too.

Continuing discussion of this on a closed bug is probably not the greatest idea. Do you want to move discussion to an issues in your project github repo, if you have one? If so, just create a bug and ref me in it.

Otherwise, we can also start a thread on my forum at https://metafetish.club, or we've got a discord server with development channels that I'm pretty much always on at https://discord.buttplug.io.

commit 9f603e33 (2018-05-19): Add lovense type/letter matching table

Fixes #18