buttplugio/stpihkal #39: Add documentation for MOSADI/CSP prostate massager

qdot, 2018-06-24:


blackspherefollower, 2018-06-24:
Incomplete knowledge dump (will update as I progress)

App 4fun-cst: https://apkgk.com/com.fun4.hackbuteer/download
jdoe12991@gmail.com
4funPassword

Commands sent have command codes followed by command specific data.

Suspend(-77): 0xB3 0xXX (where XX is the mode. Only 06 or 0C observed, although 0-12 seem valid)

Vibrate?(-73): 0xB7 0xXX 0xYY 0x??...
0xXX is unknown, but seems to be 6 or 12,
0xYY is the pattern line number 0-254?

The result looks like:
b7:06:00:70:64:00:00:00:00:00:70:64:00:00:00:00:0a
b7:06:01:70:64:70:9e:00:00:14:70:64:00:00:00:00:1e
b7:06:02:70:64:00:00:00:00:28:70:64:00:00:00:00:32
b7:06:03:70:64:00:00:00:00:3c:70:64:00:00:00:00:46
b7:06:04:70:64:00:00:00:00:50:70:64:00:00:00:00:5a
b7:06:ff (See VibrateEnd?)

VibrateEnd?(-73): 0xB7 0xXX 0xFF (XX seems to be either 6 or 12)

Unknowns (not a complete list):
0: 0x00 0xXX (where XX in the range 1-4)
-80: 0xB0 0xXX (where XX in the range 0-12)
-79: 0xB1 0xXX (where XX in the range 0-12)
-78: 0xB2 0xXX (where XX in the range 0-11, but seen to use 6 or 12)
-76: 0xB4 0xXX (where XX in the range 0-11)
-75: 0xB5 0xXX (where XX in the range 0-12)
-74: 0xB6 0xXX (where XX in the range 0-11)
-112: 0x90 0xXX (where XX in the range 0-1)
-111: 0x91 0xXX (where XX in the range 0-1)
-110: 0x92 0xXX 0xYY
-109: 0x93 0xXX (where XX in the range 0-1)
-108: 0x94 0xXX (where XX in the range 0-100)
-107: 0x95 0x01
-96: 0xA0 0xXX 0x00 0x01 0xXX 0x00 0x01 0xXX 0x00 0x01 (yes, the 0xXX 0x00 0x01 repeated 3x)
-95: 0xA1 0xWW 0xXX 0xYY 0xZZ
-94: 0xA2 0xXX

blackspherefollower, 2018-06-24:
Additionally, here's an Amazon.de link: https://www.amazon.de/dp/B0734TCG2L

blackspherefollower, 2018-06-25:
```
2018-06-25 15:07:24.5632|INFO|UWPBluetoothManager|Device: ihole
Service: 00001800-0000-1000-8000-00805f9b34fb (0x0001)
	Characteristic: 00002a00-0000-1000-8000-00805f9b34fb (0x0002): Read
	Characteristic: 00002a01-0000-1000-8000-00805f9b34fb (0x0004): Read
	Characteristic: 00002a02-0000-1000-8000-00805f9b34fb (0x0006): 10
	Characteristic: 00002a03-0000-1000-8000-00805f9b34fb (0x0008): 10
	Characteristic: 00002a04-0000-1000-8000-00805f9b34fb (0x000A): Read
Service: 00001801-0000-1000-8000-00805f9b34fb (0x000C)
	Characteristic: 00002a05-0000-1000-8000-00805f9b34fb (0x000D): Indicate
Service: 0000180a-0000-1000-8000-00805f9b34fb (0x0010)
	Characteristic: 00002a23-0000-1000-8000-00805f9b34fb (0x0011): Read
	Characteristic: 00002a24-0000-1000-8000-00805f9b34fb (0x0013): Read
	Characteristic: 00002a25-0000-1000-8000-00805f9b34fb (0x0015): Read
	Characteristic: 00002a26-0000-1000-8000-00805f9b34fb (0x0017): Read
	Characteristic: 00002a27-0000-1000-8000-00805f9b34fb (0x0019): Read
	Characteristic: 00002a28-0000-1000-8000-00805f9b34fb (0x001B): Read
	Characteristic: 00002a29-0000-1000-8000-00805f9b34fb (0x001D): Read
	Characteristic: 00002a2a-0000-1000-8000-00805f9b34fb (0x001F): Read
	Characteristic: 00002a50-0000-1000-8000-00805f9b34fb (0x0021): Read
Service: 0000180f-0000-1000-8000-00805f9b34fb (0x0023)
	Characteristic: 00002a19-0000-1000-8000-00805f9b34fb (0x0024): 18
Service: 0000ffe0-0000-1000-8000-00805f9b34fb (0x0028)
	Characteristic: 0000ffe1-0000-1000-8000-00805f9b34fb (0x0029): Notify
Service: 0000fff0-0000-1000-8000-00805f9b34fb (0x002D)
	Characteristic: 0000fff3-0000-1000-8000-00805f9b34fb (0x002E): 12
	Characteristic: 0000fff4-0000-1000-8000-00805f9b34fb (0x0031): Notify
Service: f000ffc0-0451-4000-b000-000000000000 (0x0035)
	Characteristic: f000ffc1-0451-4000-b000-000000000000 (0x0036): 28
	Characteristic: f000ffc2-0451-4000-b000-000000000000 (0x003A): 28
```

blackspherefollower, 2018-06-26:
CST products:

hiss-remi, 2018-07-01:
CST products seem to be hard to find at this point. (At least I've had no luck finding anyplace with the hackbuteer in stock that ships to the US.)

donnerruebe, 2018-07-11:
Here is the Startup of direct control...

`153 *   ->  ihole 0x00000052 93:00  undefined
155 *   ->  ihole 0x00000052 a0:0b:00:01:0b:00:01:0b:00:01  undefined
156 *   ->  ihole 0x00000052 80:04  undefined
157 *   ->  ihole 0x00000052 a1:00:14:00:01  undefined
158 *   ->  ihole 0x00000052 a1:01:14:00:01  undefined
159 *   ->  ihole 0x00000052 a1:02:14:00:01  undefined
160   ->  ihole 0x00000012 undefined  undefined
164 ihole  ->   0x00000013 undefined  undefined
165 *   ->  ihole 0x00000052 a0:0b:00:01:0b:00:01:0b:00:01  undefined
167 *   ->  ihole 0x00000052 a0:0b:00:01:0b:00:01:0b:00:01  undefined
168 *   ->  ihole 0x00000052 a0:0b:00:01:0b:00:01:0b:00:01  undefined
170 *   ->  ihole 0x00000052 a0:0b:00:01:0b:00:01:0b:00:01  undefined
171 *   ->  ihole 0x00000052 a0:0c:00:01:0c:00:01:0c:00:01  undefined
173 *   ->  ihole 0x00000052 a0:0c:00:01:0c:00:01:0c:00:01  undefined
175 *   ->  ihole 0x00000052 a0:0e:00:01:0e:00:01:0e:00:01  undefined
176 *   ->  ihole 0x00000052 a0:11:00:01:11:00:01:11:00:01  undefined
178 *   ->  ihole 0x00000052 a0:15:00:01:15:00:01:15:00:01  undefined
179 *   ->  ihole 0x00000052 a0:1a:00:01:1a:00:01:1a:00:01  undefined
181 *   ->  ihole 0x00000052 a0:1f:00:01:1f:00:01:1f:00:01  undefined
`

donnerruebe, 2018-07-12:
NOBLE about Services and Characteristics:

```
peripheral with ID 1893d72610bc found
  Local Name        = ihole
  Service Data      = []
  Service UUIDs     = ffe0,fff0,180f

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       69686f6c65 | 'ihole'
  2a01 (Appearance)
    properties  read
    value       0000 | ''
  2a02 (Peripheral Privacy Flag)
    properties  read, write
    value       00 | ''
  2a03 (Reconnection Address)
    properties  read, write
    value       000000000000 | ''
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       5000a0000000e803 | 'P h'
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
180a (Device Information)
  2a23 (System ID)
    properties  read
    value       bc10260000d79318 | '<&W'
  2a24 (Model Number String)
    properties  read
    value       4353543035303100 | 'CST0501'
  2a25 (Serial Number String)
    properties  read
    value       4d494400 | 'MID'
  2a26 (Firmware Revision String)
    properties  read
    value       323031373035303900 | '20170509'
  2a27 (Hardware Revision String)
    properties  read
    value       313031302d322d322d3600 | '1010-2-2-6'
  2a28 (Software Revision String)
    properties  read
    value       312e3100 | '1.1'
  2a29 (Manufacturer Name String)
    properties  read
    value       43535400 | 'CST'
  2a2a (IEEE 11073-20601 Regulatory Certification Data List)
    properties  read
    value       fe006368756e7368756974616e67 | '~chunshuitang'
  2a50 (PnP ID)
    properties  read
' value       010d0000001001 | '
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       3e | '>'
ffe0
  ffe1 (Key Press State)
    properties  notify
fff0
  fff3 (data output )
    properties  writeWithoutResponse, write
  fff4 ( data input )
    properties  notify
f000ffc004514000b000000000000000
  f000ffc104514000b000000000000000 (Img Identify)
    properties  writeWithoutResponse, write, notify
  f000ffc204514000b000000000000000 (Img Block)
    properties  writeWithoutResponse, write, notify
```

donnerruebe, 2018-07-12:
Got IT!!!

send
```
 93:00  
a0:0b:00:01:0b:00:01:0b:00:01  
80:04  
a1:00:14:00:01  
a1:01:14:00:01  
a1:02:14:00:01  
```
to characteristic 0xfff3

```
a0:MM:00:01:NN:00:01:PP:00:01 
```
sets the speeds for Motor 1:MM , 2:NN , 3:PP (must be send periodicly (ca. 100ms))