buttplugio/stpihkal #63: Possible change in WeVibe Protocol with Vector

qdot, 2019-08-01:
Users of the Vector are complaining that, while the device is found, control only has one speed. Anything above zero is just "on".

cc @blackspherefollower

blackspherefollower, 2019-09-06:
```
    "requires8BitIntensity": true,
```
I guess the vector has 4bits more range than the prior models.

The Moxie and Melt seem to be the same. The Melt maxing out at 25 apparently.

blackspherefollower, 2021-10-06:
Just spotted this:
```javascript
e.pairDevice = function(t) {
    return n.default.async(function(c) {
        for (;;) switch (c.prev = c.next) {
            case 0:
                return c.next = 2, n.default.awrap(u.default.write(t, "F000BB03-0451-4000-B000-000000000000", "F000C000-0451-4000-B000-000000000000", [15, 3, 0, 0, 0, 0, 0, 0]).catch(function() {}));
            case 2:
            case "end":
                return c.stop()
        }
    }, null, null, null, Promise)
}
```