buttplugio/stpihkal #67: Document Vibease Protocol

qdot, 2019-09-01:
Document protocol for Vibease hardware

eldstal, 2019-09-01:
## BLE Packet encryption

Packets are encrypted using a basic offset-by-one-and-xor method.

There are two hardcoded keys in the vibease.bluetoothtest app, as well as support for a third, possibly dynamic key.

```python
KEY1 = "2iYNPjW9ptZj6L7snPfPWIH5onzQ0V1p".encode("ascii")
KEY2 = "4sRewsha3G54ZqEcjr9Iadexd1sKB8vr".encode("ascii")
```

Here is a python implementation of the function I've chosen to call `Scramble` as well as its complement `Descramble`

```python
# Pass in a byte array cryptext
# and a byte-array for a key
# Returns a string
# This is just a plain xor cipher with an offset by one. No big deal.
def Descramble(cryptext, key):
  plaintext = [ b for b in cryptext ]

  for i in range(len(plaintext)):
    plaintext[i] = (plaintext[i] - 1) ^ key[i % len(key)]

  return bytes(plaintext).decode("ascii")

# This is the complementary function, which always uses the third key
# and a byte-array for a key
# Returns a byte array
def Scramble(plaintext, key):
  cryptext = [ b for b in plaintext.encode("ascii") ]

  for i in range(len(cryptext)):
    cryptext[i] = (cryptext[i] ^ key[i % len(key)]) + 1

  return bytes(cryptext)
```

## BLE Packet fragmentation
Payloads are scrambled according to the above and then fragmented if necessary.
A long scrambled payload is split into 16-byte chunks which are surrounded with ASCII markers.
With data bytes denoted as  `DDDD...`, here are the various packet formats I've seen so far:

* A single-chunk payload of data is transmitted as `*DDDDD!`
* A multi-chunk payload:
  * First 16-byte chunk: `*DDDDDDDDDDDDDDDD>`
  * Following 16-byte chunk(s): `<DDDDDDDDDDDDDDDD>`
  * Last 16-byte chunk: `<DDDDDDD!`

Each chunk is transmitted as an ATT packet.

Here is a python implementation of the BLE scrambling and fragmentation employed in vibease.bluetoothtest:

```python
# Pass in a string, get a list of strings for data packets back
# This is how the app breaks a longer payload up for transmission
# in short BLE packets
def ScrambleAndFragment(payload, key):
  scrambled = Scramble(payload,key).decode("ascii").replace("\n", "")
  n_blocks = int(len(scrambled) / 16)
  if (len(scrambled) % 16 != 0):
    n_blocks += 1

  if (n_blocks == 1):
    # Single packet
    return [ "*" + scrambled + "!" ]

  packets = [ ]
  for b in range(n_blocks):
    chunk = scrambled[b*16:(b+1)*16]
    if (b == 0):
      # First packet
      packets += [ "*" + chunk + ">" ]
    elif (b == n_blocks - 1):
      # Last packet
      packets += [ "<" + chunk + "!" ]
    else:
      # Middle packets
      packets += [ "<" + chunk + ">" ]

  return packets
```

The code of the app works like this (scramble, then fragment).
I have dumped some packets that indicate another mode of base64 followed by fragmentation. I have yet to properly descramble or parse any payloads, so I'm not yet sure if there is a scrambling step involved as well.

Some interesting BLE dumps are [here](https://github.com/eldstal/vibease/tree/master/dumps).

eldstal, 2019-09-01:
## UUIDs of note
The following Bluetooth UUIDs are referenced, but I have not yet determined their purpose:
```
00002902-0000-1000-8000-00805f9b34fb
0002a4d-0000-1000-8000-00805f9b34fb
803C3B1F-D300-1120-0530-33A62B7838C9
DE3A0001-7100-57EF-9190-F1BE84232730
0002a4d-0000-1000-8000-00805f9b34fb
803C3B1F-D300-1120-0530-33A62B7838C9
DE3A0001-7100-57EF-9190-F1BE84232730
00002902-0000-1000-8000-00805f9b34fb
```

eldstal, 2019-09-02:
## The third encryption key
The third key is sent from the vibrator and used in the host.
Observe the three packets numbered 1029, 1031, 1032 in [this dump](https://github.com/eldstal/vibease/blob/master/dumps/testapp_pair_and_vibrate.log)

The host sends a write to UUID `803C3B1F-D300-1120-0530-33A62B7838C9` with the contents `$aGk=!` and the response comes back fragmented in two packets.
```
#fSFwIxA6Oy9VNAJTNS>
<ECNixC!
```
Strip out the delimiting characters and concatenate to get the intended payload `fSFwIxA6Oy9VNAJTNSECNixC`.

Base64-decode this and `Descramble()` using the (hardcoded) KEY2 above:
`HS=GxJROgt4fnQDVA3`

From debugging the application, the key should indeed be `GxJROgt4fnQDVA3`.
Due to this label, I've chosen to call this the HS key.

I have not seen this key change between sessions, so it is possible that it is a static function of either the vibrator's unique IDs, the host (my phone) or both.

eldstal, 2019-09-04:
## The initial handshake

I can confirm that the HS key is accessible by following these steps:

```
1. Open a GATT connection to the vibease
2. Locate service de3a0001-7100-57ef-9190-f1be84232730
3. Find the characteristic 803C3B1F-D300-1120-0530-33A62B7838C9
   * There are two of them, one with the READ property and one with the
     WRITE_WITHOUT_RESPONSE property. I will call them cmd_read and cmd_write.
4. Enable change notification on cmd_read, so that you receive messages when the value changes.
5. Perform a write to cmd_write with the payload $aGk=!   (bytes [ 0x24, 0x61, 0x47, 0x6b, 0x3d, 0x21 ])
6. Three notifications come in, in order. In my setup they are:
   #fSFwIxA6Oy9VNAJTNS>
   <ECNixC!
   %1406-OK!
```

This lets us retrieve the HS key and it looks like the device confirms a completed handshake, or similar.

It appears from the code that KEY2 is always used to descramble messages FROM the vibease, and except for this initial handshake, KEY_HS is always used to scramble messages TO the vibease.

eldstal, 2019-09-06:
There is a slight inconsistency in the `Scramble()` with KEY_HS. Instead of using `len(key)` for the modulo, it uses `len(key-1)` which means the last character of KEY_HS is never actually used. This 

With the python example above, it is sufficient to truncate the very last byte of KEY_HS and both `Scramble()` and `Descramble()` should work as expected using all known keys.

This way, the same function can descramble packets transmitted by the host (using the truncated KEY_HS) and packets transmitted by the device (using the full KEY2).

eldstal, 2019-09-06:
## Known commands
(The examples below use my `KEY_HS=GxJROgt4fnQDVA` and will look different on a different setup.)

### Vibrate
Unscrambled example: `1200,2200,3200,4200,5200,6200,7200,8200,9200,0200`
Transmitted packets
```
*d0t7Y2RWRwVXQ2N3>
<Z3JsTXljgExCB1df>
<fnNlcnhVfmGAWFkN>
<VV9iaXB0eElnY35Y>
<RQ==!
```
Notes: The first digit is probably intensity while the remaining three could be duration (ms).

### Stop
Unscrambled example: `0500,0500`
Transmitted packets:
```
*eE57Y2RYQgVX!
```

eldstal, 2019-09-07:
## Message prefixes
### Host -> Device
`$` appears to signify that the message is scrambled using KEY2 rather than KEY_HS
`*` appears to signify normal commands, scrambled with KEY_HS

### Device -> Host
`!` is used for some sort of version packet, which is entirely unscrambled and not b64 encoded
`#` appears to be used for normal responses, scrambled with KEY1

eldstal, 2019-09-08:
## HELO
After key exchange has been completed, the device sends an unscrambled and un-base64'd message like
```
%1406-OK!
```

which appears to be a version number and a basic status report.