buttplugio/stpihkal #68: Add Zalo Queen documentation

blackspherefollower, 2019-09-02:
BLE name: ZALO-Queen
Servive UUID: 0000fff0-0000-1000-8000-00805f9b34fb
Tx Char UUID: 0000fff1-0000-1000-8000-00805f9b34fb

Tx prorocol: 0xAA 0xBB 0xCC 
AA is mode:
- 01 is manual control
  BB controls the oscillator at the tip (values 01-08, 01 is off)
  CC controls the vibrator (values 01-08, 01 is low, no off)
- 02 is off
  BB and CC should be 01, but doesn't seem to matter
- 03 is pattern:
  BB is the pattern (01-08)
  CC is the intensity (01-08)

I suspect that the same protocol applies for all Zalo devices, but I only have the Queen.

TheHardew, 2021-01-28:
BLE name: ZALO-King
Servive UUID: 0000fff0-0000-1000-8000-00805f9b34fb
Tx Char UUID: 0000fff1-0000-1000-8000-00805f9b34fb

Tx prorocol: 0xAA 0xBB 0xCC
AA is mode:

 - 01 is manual control
BB controls the vibrator (values 01-04, no off)
CC controls the oscillator at the tip (values 01-08, no off)
(it is reversed wrt the Queen)
 - 02 is stop
BB controls the vibrator
CC controls the oscillator
if BB or CC are 01, the respective function will be turned off
 - 03 is pattern:
BB is the pattern (01-08)
CC is the intensity (01-08)
 - 04 is turn off the device
 BB and CC don't matter and can even not be sent

You can turn on just the oscillator by e.g. first sending `0x01 0x01 0x01` and then `0x02 0x01 0x00`. I don't know if there's any other option.

I'm also ataching a log of all the things I was able to query via D-Bus.
[ZALO-King.log](https://github.com/buttplugio/stpihkal/files/5888855/ZALO-King.log)

blackspherefollower, 2021-07-05:
BLE name: ZALO-Jeanne
Servive UUID: 0000fff0-0000-1000-8000-00805f9b34fb
Tx Char UUID: 0000fff1-0000-1000-8000-00805f9b34fb

Tx protocol: 0xAA 0xBB 0xCC
AA is mode:

    01 is manual control
    BB controls the vibrator at the tip (values 01-08, 01 is off)
    CC unused (values 01-08, 01 is low, no off)
    02 is off
    BB and CC should be 01, but doesn't seem to matter
    03 is pattern:
    BB is the pattern (01-08)
    CC is the intensity (01-08)

blackspherefollower, 2021-07-12:
@TheHardew I picked up a King... because I have issues... (See https://github.com/blackspherefollower/IoSTIndex/issues/716)

I did some testing, and found that the vibrator on the King has a district tone difference between 4 and 8 (although the vibe's relativity weak so might not be noticeable to a real user).

I think this probably makes the protocol easier to document, since the 1 vibe and multi-feature devices have consistent control commands.

closed 2026-03-01 by qdot