buttplugio/stpihkal #70: Update Lovense protocol documentation

qdot, 2019-09-16:
Add information about Vibrate[x] indexes, patterns, etc...

qdot, 2020-07-19:
Also need to add:

- Alarm Commands ("AI", "AN", etc)
- Light Commands ("GetLights")
- BT commands?

geopard1983, 2021-08-06:
-------------------------------------------------------------------------------
how you should send commands to buttplug device (directly):
- don't send it immediately
- feed command to command queue
- make worker that gets one command from command queue
and sends it every 100 miliseconds
-------------------------------------------------------------------------------
List of devices that support PIN 
(firmware is second paremeter returned by "DeviceType;" command)
-------------------------------------------------------------------------------

Code Letter, Name, Firmware range (inclusive)

```
R     Diamo      3..3
S     Lush       not supported
C/A   Nora       139..199
B     Max        106..129 and 132..199 and 206..209 and 215..299
Z     Hush       205..219 and 224..299
L     Ambi       104..199
P     Edge       105..199 and 203..299
W     Domi       106..199
O     Osci       208..299
X     Ferri      3..99
V     Mission    not supported
```

Pin operations:
----------------------------------------------------------------------------
(if device is not Osci, or firmware version is not 206, messages should be
named SetPinS, SetPinC instead of SetPinStatus, SetPinCode;
GetPinS remains the same for both)
----------------------------------------------------------------------------

- To get pin status, send `"GetPinS,gp;"`, result should be `"pinstatus:1"` or
`"pinstatus:0"`

- To disable pin, send `"SetPinStatus:0,so;"`, wait 2000 miliseconds (2 seconds),
and send `"PowerOff;"`

- To (just) enable pin, send `"SetPinStatus:1,sf;"`

- To set pin:

\<PIN\> - this value should be 6 digits, please don't try anything else than 0..9

Send `"SetPinCode:\<PIN\>,sc;"`, if result is `"OK;"` wait 2000 miliseconds,
then send `"SetPinStatus:1,sf;"`, if result is `"OK;"` wait 1500 miliseconds, 
then send "PowerOff;"

Pattern operations:

It's not clear to me how many patterns each device supports, I know for sure
Edge 2 (Edge) has 10.
Pattern indexes start at 1.
\<INDEX\> is index of pattern (don't use < >, those are just for clarity)
\<DATA\> is data of pattern (for example 00050005)
\<CHUNK_N\> is chunked data of pattern (explained later)
\<CNT\> - count (number) of ALL chunks

- To delete pattern, send "Remove:\<INDEX\>,d\<INDEX\>;"

- To set pattern:

If device is NOT Domi or or firmware version is greater than 37,
\<CHUNKSIZE\> should be 9, or 5 otherwise.

If pattern is 1 or 2 digits long, just send:
`"Ps:\<INDEX\>:\<DATA\>,w1;"`, then wait 160 miliseconds.

(unconfirmed: you should get "OK,w1;" ?)

If pattern is longer than to digits, divide it in \<CHUNKSIZE\> chunks,
where number of ALL chunks will be \<CNT\>.

\<I_OR_f\> is number of loop passes or letter "f" if \<I\> > 100

WARNING: if device IS Domi and firmware version is smaller than 38,
you should send `"Pl<INDEX>:<I>/<CNT>:<CHUNK_I>;"` (without `",w<I_OR_f>"`)

Make loop:
	from CHUNK_1 to CHUNK_\<CNT\>, where \<I\> is number of passes (starting from 1,
increasing up to \<CNT\>, for C programmers: `for (int \<I\>=1; \<I\> <= \<CNT\>, <I>++)`;
for python coders: `for \<I\> in range(1, \<CNT\>+1)`:

	- send `"Pl<INDEX>:<I>/<CNT>:<CHUNK_I>,w<I_OR_f>;"`
	- wait 160 miliseconds
    - (unconfirmed: you should get "OK,w<I>;" ?)

(END LOOP)

EXAMPLE:
if pattern is "000001111122222333334444455555", we divide into 9-character
long chunks:

1. "000001111"
2. "122222333"
3. "334444455"
4. "555"

Then we send (assuming our pattern index in the device is 7):
	`"Pl7:1/4:000001111,w1;"`
	`"Pl7:2/4:122222333,w2;"`
	`"Pl7:3/4:334444455,w3;"`
	`"Pl7:4/4:555,w4;"`

(we should wait at least 160 miliseconds after each send).

blackspherefollower, 2022-06-11:
Lovense Exomoon info:
Identifier: EF
Service UUID: 45460001-0023-4bd4-bbd5-a6920e4c5653
Tx: 45460002-0023-4bd4-bbd5-a6920e4c5653
Rx: 45460003-0023-4bd4-bbd5-a6920e4c5653