buttplugio/stpihkal #73: Need more detail on KiiRoo "Get Software Version" command

Prof62, 2019-10-05:
The command response for KiiRoo firmware command 0x05 shows where the software version number is in the response but does not specify the purpose of the other 4 (or 10) bytes.

These seem to have some significance because, although the iOS Feel Connect app shows a simulated Launch as being online, it flags it as having an unknown software version and doesn't send any further data to it.

Can anyone provide either an example of the full response to a 0x05 command from a Launch or details of the other fields so that I can  cook up a response that the app will accept.

qdot, 2019-10-05:
Unfortunately the only time I've worked with that packet is in the firmware uploader, and that was just for version identification.

https://github.com/metafetish/kiiroo-firmware-loader/blob/master/src/bluetooth.ts#L99

qdot, 2019-10-05:
Note the Ghidra does have the ability to disassemble the firmware, but I never really dug that far into it due to lack of time.

airzocker, 2021-07-30:
@Prof62 @qdot  Hi, I am trying to simulate a Launch as well (on ESP32) but I just can't get it to show up in Feel Connect, is there anything else it needs other than name "Launch" and the service with 3 read/write characteristics all with the right uuids (https://stpihkal.docs.buttplug.io/protocols/fleshlight-launch.html#launch-protocol)? It can't be the problem with my implementation because the app doesn't even write anything to any of the characteristics...