buttplugio/stpihkal #87: Document Vorze Piston SA protocol

qdot, 2020-02-14:
Bluetooth Name: VorzePiston

Protocol is same as Cyclone and UFO SA. 3 bytes, identifier value for the Piston is 3.

This means the commands are in the following format:

```
[0x3, 0x1, 0xZZ]
```

0x1 in the byte 1 position is used for rotation speed and direction in the UFO and Cyclone (0x3 is vibrate for the Bach Smart). So I'm not sure if 0xZZ here is setting some sort of stroke speed, or a position. I would guess it's stroke speed.

blackspherefollower, 2020-04-23:
According to https://github.com/amateras977/A10ServerBLE/blob/ddc8c7fb77b889f644377412cadf7322901e2ddb/A10ServerBLE/TargetDevice/VorzeA10Piston.cs

It looks like an initialisation message of `[0x03, 0x01, 0x00]` is required.

After that messages of the form `[0x03, 0xXX, 0xYY]` are sent to change the position:
* 0xXX is speed (range 10-60 or 0x0A-0x3C in hex)
  Forward and reverse speeds are not equal
* 0xYY is position (range 0-200 or 0x00-0xC8 in hex)
  Position 0 is the front (need to test but I think this means the carriage is closest to the opening)

The code states that the device ignores messages sent within 20 milliseconds of the prior message. I'm surprised that the BLE stack can even go that fast.

I wonder what the appropriate language is to describe position in real terms: this device alters its cavity depth, so at position 200 it's at it's deepest. But that's a side effect of the onahole's built-in stretching guard(?). The SayberX has a similar design, which I guess focuses the frictions on the assumed nerve center placed a few inches within the devices cavity, avoiding the risk if pinches.
The launch doesn't have an equivalent guard, but I don't think it's intended to move the fleshlight so far that the nerve center leaves the onahole. This concludes my weird segway into semantics of insertable terminology.

Once I get some time, I'll write some code which moves the device over various ranges at various speeds to see if I can map the speed/distance/direction/time by recording audio.

Nootna8, 2021-07-01:
[VORZE_SDK.pdf](https://github.com/buttplugio/stpihkal/files/6749833/VORZE_SDK.pdf)
Was asked to post this here

qdot, 2021-07-01:
> 
> 
> [VORZE_SDK.pdf](https://github.com/buttplugio/stpihkal/files/6749833/VORZE_SDK.pdf)
> Was asked to post this here

Woah, this is awesome, thanks! :D

Nootna8, 2021-07-01:
Found eng version as well
[VORZE_SDK_en.pdf](https://github.com/buttplugio/stpihkal/files/6750156/VORZE_SDK_en.pdf)