buttplugio/stpihkal #99: Document Satisfyer Protocol

XDjackieXD, 2020-09-15:
Here the information I found out about the Satisfyer devices (Specifically the Royal One):

If the device was paired to a host before, it has to be reset by holding the button for 15s (it should do 5 fast vibrations once the button has been held long enough).
After that it can be made visible by holding the button for 4s (2 fast vibrations).
It doesn't send a name in the BLE advertisement packets (or I couldn't find it). Only *after* the first connection the host knows the name (seems to be GATT attribute device name).

The Service of interest is "51361500-c5e7-47c7-8a6e-47ebc99d80e8" ("live control" according to the android app code).
It exposes two Characteristics:
- "51361501-c5e7-47c7-8a6e-47ebc99d80e8" (motor mode): This has to be set to 0x01 ("start") before the motor value can be set. The app also knows the values 0x03 ("pause") and 0x07 ("stop"). Both seem to stop the vibration and require a "start" again and I couldn't figure out the difference.
- "51361502-c5e7-47c7-8a6e-47ebc99d80e8" (motor value): This accepts a range of 0x00000000 to 0x00000064 (yes it has to be 4 bytes long with the least significant byte from 0 to decimal 100).

When the motor value is set to 0 for a few seconds it seems to exit "start" mode and requires the start message to be sent again.

Standard ble services it exposes:
- battery level service (0000180f-0000-1000-8000-00805f9b34fb)
  - battery level characteristic (00002a19-0000-1000-8000-00805f9b34fb)
- device information service (0000180a-0000-1000-8000-00805f9b34fb)
  - manufacturer name string (00002a29-0000-1000-8000-00805f9b34fb): "Concepter"
  - model number string (00002a24-0000-1000-8000-00805f9b34fb): ""
  - serial number string (00002a25-0000-1000-8000-00805f9b34fb): ""
  - hardware revision string (00002a27-0000-1000-8000-00805f9b34fb): "0.0.1"
  - firmware revision string (00002a26-0000-1000-8000-00805f9b34fb): ""
  - software revision string (00002a28-0000-1000-8000-00805f9b34fb): "0.0.1"

Other exposed vendor specific Services that I haven't figured out the usage yet are
- "51361600-c5e7-47c7-8a6e-47ebc99d80e8" ("device control")
  - "51361601-c5e7-47c7-8a6e-47ebc99d80e8" ("control mode")
  - "51361602-c5e7-47c7-8a6e-47ebc99d80e8" ("control value")
  - "51361603-c5e7-47c7-8a6e-47ebc99d80e8" ("rename value")
- "51361700-c5e7-47c7-8a6e-47ebc99d80e8" ("device info")
  - "51361701-c5e7-47c7-8a6e-47ebc99d80e8" ("hardware info")

XDjackieXD, 2020-10-23:
The Satisfyer Love Triangle works exactly the same way but the "51361502-c5e7-47c7-8a6e-47ebc99d80e8" (motor value) attribute now expects 8 bytes of data where the lower 4 bytes represent the suction motor and the higher 4 bytes the vibration motor.
for example 0x0000006400000000 turns on the vibration motor to max (decimal 100) and the suction off.
0x0000000000000040 turns on the suction device to 64% but the vibration motor off.

denialtek, 2020-11-13:
I had to write the same value to each of the 4 bytes or else the vibrator/suction would stutter off and on quickly after each adjustment (testing on the Satisfyer Curvy 3).
ex. 0x6464646400000000  or 0x0000000040404040

XDjackieXD, 2020-11-13:
I thought so too at first but playing around with it suffices to always send the correct amount of bytes (4 Bytes per motor) and have the LSB be the motor value. at least that works 100% reliably for both satisfyer toys I could try it with (tested with chromium ble tools and the nrf ble app)

XDjackieXD, 2021-08-29:
@denialtek  I must have done something wrong during my initial testing but I can reproduce your stutter problem now (with an initial implementation in buttplug-rs by @blackspherefollower).
I suspect that the four bytes are some form of timed sequence to reduce the overhead by sending 4 values of a vibration pattern per packet but I haven't had time to check this further (I could not get it to a point where I could feel the 4 values consecutively so I'd probably have to check with an accelerometer and a scope or something like that ^^)

liamdm, 2022-02-21:
Last bit controls the target speed of the motor, the first three bytes are used to control what happens to the motor while it's transitioning to the next speed from what I can tell. If you set the first three bits to the same value as the target speed, the transition will be smooth (no stuttering) between commands. But if you are sending commands quickly together (eg 100ms delay), say there is a 3ms transition between motor speeds, the first bit seems to be the speed target for the first 1ms of the transition, the second for the 2ms and third for 3ms. It's not enough time for the motor to respond reliably but it adds a bit of jerk and additional jitter (probably used for more fine control and responsiveness with the touch input from the app). You can notice this if you set the bits to (100, 100, 100, 50) and then send one command every second, you will notice there is a bit of additional jitter every second, whereas if you set it to (50, 50, 50, 50) this jerk is gone. This is probably used to achieve a "jerk" in the motor which I think makes the vibration different than if the motor had a linear speed transition. 

As for the 600 range UUID's, there are some more in the newer version:
```
private static final UUID alarm =               UUID.fromString("51361603-c5e7-47c7-8a6e-47ebc99d80e8");
private static final UUID heating =             UUID.fromString("51361604-c5e7-47c7-8a6e-47ebc99d80e8");
private static final UUID keepAliveControl =    UUID.fromString("51361605-c5e7-47c7-8a6e-47ebc99d80e8");
private static final UUID pumpControl =         UUID.fromString("51361606-c5e7-47c7-8a6e-47ebc99d80e8");
private static final UUID lockControl =         UUID.fromString("51361607-c5e7-47c7-8a6e-47ebc99d80e8");
```

blackspherefollower, 2022-03-01:
@liamdm Which device were you testing with?

I've just received a "Heat Climax" which appears to send a heat command of 0x01 0x28 0xff (which then gets updates from the device with the 3rd byte seemly reflecting the actual temperature).

The 4-byte motor control is common across most of the Satisfyer range where either 1 and 2 motors are used: I suspect that you're using a single motor device and the first 2 bytes are unused by your device.

liamdm, 2022-03-01:
So the heat command to control the heat via the heating characteristic I observed was only a single byte value (possibly to set the state). I get the feeling that 0x28 is a temporary state that is "warming up" because the actual constant I saw in the APK code was different, but I have verified this working on my heat climax as being able to turn the heating on and off by sending that. 

As for the motor control, agreed, it might be due to it being a single motor device, as I was also testing this on a heat climax (have a three motor device now though and want to check this too) - but the first and second byte definitely had an impact (on the transition speed). Maybe this is some quirk of how the single motor devices handles these values. If you'd like I can send through some java code that you can build into an Android app to test this?

blackspherefollower, 2022-03-01:
@liamdm I suspect the original Satisfyer devices had the same intermediate speed value, but just one byte: [ motor2-intermediate, motor2-final, motor1-intermediate, motor1-final ] and the newer have 4-bytes per motor, with a variable length data field.

I generally test with the nRF Connect app (really easy to send/receive packets on mobile. And capture data by analysing the btsnoop_hci.log, but I will take a look at the APK to see if there's a clear firmware based protocol switch.

blackspherefollower, 2022-03-01:
Looks like the heatbeat was added in fw2.0.24 and keepalive was added in fw2.0.34 (not sure what either do).

It looks like the heater result is [ boolean heating, int targetTemp, int currentTemp ]. I must have misread my wireshark trace: I only see 01 or 00 sent to that characteristic.

And it looks like the vibration commands haven't changed, my memory was just playing tricks on me. Looks like ALL Satisfyer devices can be controlled by 4-byte per vibe (4 for 1 vibe, 8 for 2, 12 for 3).

Furthermore, we can identify the devices prior to connection by checking the manufacturer advertisement data (I've previously pushed back on this approach since I couldn't see a way to map all the Satisfyer devices without buying them all, but as it turns out, products.json is right there for the reading)

blackspherefollower, 2022-04-05:
Digging further, I think the logic looks like:
```
if( firmwareVersion < 2.0.24) {
  // ping by setting last motor speed
} else if(firmwareVersion < 2.0.34) {
  // ping by setting controlMode = 87
} else {
  // no ping - keep-alive set on device?
}

if( oneOf( Sexy Secret, Little Secret, Signet Ring, Strong One, Mighty One, Powerful One, Royal One, Plug-ilicious 1, Plug-ilicious 2 ) )  {
// wearable
timeout = 28800s
} else {
timeout = backgroundMode ? 660s : 28800s
}
"51361605-c5e7-47c7-8a6e-47ebc99d80e8".write(timeout)

```

blackspherefollower, 2022-04-06:
This might only apply to live control, but motor updates are sent by the app once very .5 seconds.
Without frequent updates, the device locks you out regardless of any of the other keep-alive/heatbeat things.

The frequency seems to differ slightly between devices, but for the Plug-ilicous devices 3 seconds is not frequent enough, whilst 1 second seems to be fine.

blackspherefollower, 2022-12-07:
SO! I got a Air Pump Booty 5+ (it has a pump!)
Pump rx/tx: 51361606-c5e7-47c7-8a6e-47ebc99d80e8
Commands: 0x0000XXXX
* where XXXX is milliseconds to pump for (0x01f4 == 0.5s)
* if XXXX is 0000, the release value opens and the device deflates
* if the pressure reaches it's maximum, the pump will stop
Notification: 0xXXXX (where XXXX is the current pressure level, which seems to peak at 6404 - note the last byte seems only to be 01-05)

closed 2026-03-01 by qdot