buttplugio/stpihkal #100: Document CowGirl Protocol

blackspherefollower, 2020-09-24:
Note: WIP - lots of gaps in our understand right now

Name: THE COWGIRL
UUID (characteristic?): 0000fe01-0000-1000-8000-00805f9b34fb

Control data 0xAA 0xBB 0xCC 0xDD
Where:
* 0xAA in the range 0-2 (unknown)
* 0xBB in the range 0-6 (pattern)
* 0xCC in range 0-255 (rotation speed?)
* 0xDD in range 0-255 (vibration speed?)

blackspherefollower, 2020-09-25:
See: https://metafetish.club/t/the-cowgirl-support-coming/1408/7

denialtek, 2022-12-21:
There's a Unicorn special edition.

Name: THE UNICORN
The protocol is the same.

blackspherefollower, 2022-12-27:
@denialtek Do you know what the first byte should be for normal control?

denialtek, 2022-12-31:
I'm using 0 for the first byte and write with response.

The user I was testing with gave me a partial Bluetooth capture from the official app. In it the app always sent 0 for the first byte and 0x8c was the highest value I saw for the speed (but it's possible that's from them not actually setting the speed all the way to 100%).

closed 2026-03-01 by qdot