buttplugio/stpihkal #131: Document Fredorch F21S/T8 protocol

blackspherefollower, 2021-07-19:
BLE Name: YXlinksSPP
Service UUID: 0000ffb0-0000-1000-8000-00805f9b34fb
Tx Characteristic UUID: 0000ffb1-0000-1000-8000-00805f9b34fb
Rx Characteristic UUID: 0000ffb2-0000-1000-8000-00805f9b34fb

Device is a positional belt driven linear actuator:
* 9 oscillation patterns (speed and range controllable)
* Programmable sequences
* On-demand movements

Pattern mode:
* Initialisation: `01 06 00 64 00 00 CRC16`
* On: `01 06 00 65 00 01 CRC16`
* Off: `01 06 00 65 00 00 CRC16`
* Speed up: `01 06 00 66 00 02 CRC16`
* Speed down: `01 06 00 66 00 01 CRC16`
* Range up: `01 06 00 67 00 02 CRC16`
* Range down: `01 06 00 67 00 01 CRC16`
* Pattern select `01 06 00 68 00 0X CRC16` (X 1-9 pattern number)

Program mode:
* Initialisation: `01 06 00 64 00 01 CRC16`
* Beyond here, things get complicated and I'm not going to untangle it right now.

Free mode:
* Initialisation: `01 06 00 64 00 02 CRC16`
* Move: `01 06 00 A7 XX YY CRC16` (XX YY position 16-bit hex value but app sends 0-100 in decimal - I bet this is a bug in their app)

```javascript
_auchCRCHi = [0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64, 0, 193, 129, 64, 1, 192, 128, 65, 0, 193, 129, 64, 1, 192, 128, 65, 1, 192, 128, 65, 0, 193, 129, 64];
_auchCRCLo = [0, 192, 193, 1, 195, 3, 2, 194, 198, 6, 7, 199, 5, 197, 196, 4, 204, 12, 13, 205, 15, 207, 206, 14, 10, 202, 203, 11, 201, 9, 8, 200, 216, 24, 25, 217, 27, 219, 218, 26, 30, 222, 223, 31, 221, 29, 28, 220, 20, 212, 213, 21, 215, 23, 22, 214, 210, 18, 19, 211, 17, 209, 208, 16, 240, 48, 49, 241, 51, 243, 242, 50, 54, 246, 247, 55, 245, 53, 52, 244, 60, 252, 253, 61, 255, 63, 62, 254, 250, 58, 59, 251, 57, 249, 248, 56, 40, 232, 233, 41, 235, 43, 42, 234, 238, 46, 47, 239, 45, 237, 236, 44, 228, 36, 37, 229, 39, 231, 230, 38, 34, 226, 227, 35, 225, 33, 32, 224, 160, 96, 97, 161, 99, 163, 162, 98, 102, 166, 167, 103, 165, 101, 100, 164, 108, 172, 173, 109, 175, 111, 110, 174, 170, 106, 107, 171, 105, 169, 168, 104, 120, 184, 185, 121, 187, 123, 122, 186, 190, 126, 127, 191, 125, 189, 188, 124, 180, 116, 117, 181, 119, 183, 182, 118, 114, 178, 179, 115, 177, 113, 112, 176, 80, 144, 145, 81, 147, 83, 82, 146, 150, 86, 87, 151, 85, 149, 148, 84, 156, 92, 93, 157, 95, 159, 158, 94, 90, 154, 155, 91, 153, 89, 88, 152, 136, 72, 73, 137, 75, 139, 138, 74, 78, 142, 143, 79, 141, 77, 76, 140, 68, 132, 133, 69, 135, 71, 70, 134, 130, 66, 67, 131, 65, 129, 128, 64];

function padLeft(e, t, n) {
    void 0 == n && (n = "0");
    for (var o = 0, i = t - e.length; o < i; o++) {
        e = n + e;
    }
    return e;
}
function CRC16(t) {
    for (var n = 255, o = 255, i = 0; i < t.length; i++) {
        var a = n ^ t[i];
        n = o ^ e._auchCRCHi[a];
        o = e._auchCRCLo[a]
    }
    return [n, o];
}
```

blackspherefollower, 2021-07-22:
Since the CRC16 thing makes testing with nRF Connect hard, I threw together a test app here: https://glitch.com/edit/#!/cactus-square-debt
After connect send the messages without the CRC16 (send will calculate and add the data)

blackspherefollower, 2022-05-19:
Valid movement positions:
* 00
* 10
* 14
* 28
* 32

blackspherefollower, 2022-05-20:
Right... an update, because I've been covering my progress on my patron and I need to replicate it here.
I have a decent surface fit: TIME = 2.61607666e+03 * (SPEED ^ -9.64147031e-01) * ((DISTANCE*10) ^  2.28645432e-01)
Where time is in ms, speed is 1-15 and distance is 0-15.

Here's the relevant cross links:
* https://www.patreon.com/posts/measurements-59821917
* https://www.patreon.com/posts/continued-work-66659744
* https://www.patreon.com/posts/fredorch-curve-66668487

forest-devil, 2024-07-11:
FYI, according to the I/O byte pattern, I believe this is MODBUS RTU. You can check https://www.modbustools.com/modbus.html for more details.

Address | Function | Data | CRC
-- | -- | -- | --
1 byte | 1 byte | N bytes | 2 bytes

The first byte `02` is the device ID, the second byte is Function, where

[01 (0x01) Read Coils](https://www.modbustools.com/modbus.html#Function01)
[02 (0x02) Read Discrete Inputs](https://www.modbustools.com/modbus.html#function02)
[03 (0x03) Read Holding Registers](https://www.modbustools.com/modbus.html#function03)
[04 (0x04) Read Input Registers](https://www.modbustools.com/modbus.html#function04)
[05 (0x05) Write Single Coil](https://www.modbustools.com/modbus.html#function05)
[06 (0x06) Write Single Register](https://www.modbustools.com/modbus.html#function06)
08 (0x08) Diagnostics (Serial Line only)
11 (0x0B) Get Comm Event Counter (Serial Line only)
[15 (0x0F) Write Multiple Coils](https://www.modbustools.com/modbus.html#function15)
[16 (0x10) Write Multiple Registers](https://www.modbustools.com/modbus.html#function16)
17 (0x11) Report Server ID (Serial Line only)
22 (0x16) Mask Write Register
23 (0x17) Read/Write Multiple Registers
43 / 14 (0x2B / 0x0E) Read Device Identification

button73, 2024-07-21:
@forest-devil Do you know what is the current status of F21S support in the library? On iostindex it is added as supported, but I've seen posts on reddit that position control is not working (at least on xtoys)

blackspherefollower, 2024-07-21:
> @forest-devil Do you know what is the current status of F21S support in the library? On iostindex it is added as supported, but I've seen posts on reddit that position control is not working (at least on xtoys) 

The protocol is implemented and "works". The timings in Buttplug are off, but we do at least have the numbers (see https://www.patreon.com/posts/66668487).

Position control "should" be fine, but we do not control the Xtoys implementation.

forest-devil, 2025-03-04:
> The protocol is implemented and "works". The timings in Buttplug are off, but we do at least have the numbers (see https://www.patreon.com/posts/66668487).
>
> Position control "should" be fine, but we [...]

good enough, thanks for all your works done 👍

samsmit362, 2025-07-13:
> The protocol is implemented and "works". The timings in Buttplug are off, but we do at least have the numbers (see https://www.patreon.com/posts/66668487).
>
> Position control "should" be fine, but we [...]

What is the best curent way for acurate position control Fredorch F21S with buttplug.io.
I'm currently using [buttplugCpp](https://github.com/dumbowumbo/buttplugCpp/blob/master/example/buttplugCpp.cpp) project and already wrote [C++ program for control Hismith](https://github.com/samsmit362/HismithControl), but now going to switch to Fredorch F21S which allow also control depth.
But it is not clear what is the best way to perfome position (as result depth also), speed and etc by API.
The main idea is to make verly close complex movements which can change 2-5 times during single move in or out but for Fredorch F21S (in case of Hismith this is already working, especially on low speed movements ~1/4-2 times per sercond)

blackspherefollower, 2025-07-13:
The Buttplug implementation of the protocol allows for position control, but as in my last comment you quoted, the speed control isn't accurate.

The Buttplug protocol uses LinearCmd for linear positional devices, which is distinct from the speed only ScalarCmd oscillate command used for devices like the Hismiths

samsmit362, 2025-07-13:
> The Buttplug implementation of the protocol allows for position control, but as in my last comment you quoted, the speed control isn't accurate.
>
> The Buttplug protocol uses LinearCmd for linear [...]

Thanks, but there is still a lot of questions.
Isn't there way to use "Free mode" or "Program mode" directly? Instead of usage: LinearCmd or ScalarCmd.
How LinearCmd now is implemented, does it use "Program mode" or "Free mode" now?
Do you know where to find where F21S support now is implemented? buttplugio says that it is open source but i can't find where is code for concrete device support is in implementation.

samsmit362, 2025-07-13:
> The Buttplug implementation of the protocol allows for position control, but as in my last comment you quoted, the speed control isn't accurate.
>
> The Buttplug protocol uses LinearCmd for linear [...]

Do you have any ideas how to change speed of device like on wired controller (so that it will not wait for move finished)?

blackspherefollower, 2025-07-13:
What you see documented here is the limits of the protocol. To work out timings for speed, you have to know the prior position, which Buttplug will if you send a LinearCmd per movement: you can't change the speed during the movement.

Free mode has no speed control, which is generally a dangerous thing for a fucking machine.

The controller doesn't use the same protocol that the app does and can just change the limit values directly.

What you see here is what you get. The only improvement we're likely to make is to get the timing approximation closer.

samsmit362, 2025-07-13:
> FYI, according to the I/O byte pattern, I believe this is MODBUS RTU. You can check https://www.modbustools.com/modbus.html for more details.
>
> Address	Function	Data	CRC
> 1 byte	1 byte	N bytes	2 bytes [...]

@forest-devil  how does this connected to codes for using different modes? Can you please show where is "The first byte `02` is the device ID" in for example pattern mode code: Initialisation: 01 06 00 64 00 00 CRC16

samsmit362, 2025-07-13:
> > What you see documented here is the limits of the protocol. To work out timings for speed, you have to know the prior position, which Buttplug will if you send a LinearCmd per movement: you can't change the speed during the movement.
> > Free mode has no speed control, which is generally a dangerous thing for a fucking machine.
> > The controller doesn't use the same protocol that the app does and can just change the limit values directly.
> > What you see here is what you get. The only improvement we're likely to make is to get the timing approximation closer.
> 
> ```
> // In the protocol, we know max speed is 99
> let speed = ((message.speed() as f64 / 99.0) * 15.0) as u8;
>   let mut data: Vec<u8> = vec![
>     0x01, 0x10, 0x00, 0x6B, 0x00, 0x05, 0x0a, 0x00, speed, 0x00, speed, 0x00, position, 0x00,
>     position, 0x00, 0x01,
>   ];
> ```
> 
> [@blackspherefollower](https://github.com/blackspherefollower) doesn't this means that actual set speed values are set to 0,1,2,3-15 so have only 16 speed variations? (this looks very inaccurate case of abilities of speed control) Is there any info about acceptable speed values?

After reading [official page](https://fredorch.com.cn/products/fredorch-200w-sex-machine) it is really only 0-15 speed variations, also as for position. So it looks only free mode is acceptable for me for change pos in real time with frequency ~30 times per second.

blackspherefollower, 2025-07-13:
You'll get a max of 10 updates per second, and it'll be jerky.

Setting up a program that isn't going to oscillate only works if you move to a position (15 options) at a speed (15 options). If you want to make the device oscillate but move faster and slower per movement, you should be able to craft a program with multiple steps, but you'd have to do that yourself - you won't get support for it in Buttplug any time soon.

samsmit362, 2025-07-13:
@blackspherefollower Where do you get data for YXlinksSPP control?
I see codes for how to switch modes but don't see code how to stop device or stop executing mode, how to stop it?

blackspherefollower, 2025-07-13:
Hmm. Looks like I didn't document the stop program command, and it's been a while since I did any captures for the Fredorch.

I reversed the protocol by capturing the Bluetooth traffic.

closed 2026-03-01 by qdot