buttplugio/stpihkal #140: Document Hgod (Butterfly Love) Protocol

blackspherefollower, 2021-09-08:
BLE Name: AMN NEO
Service UUID: 0000ffe3-0000-1000-8000-00805f9b34fb
Tx Characteristic UUID: 0000ffe1-0000-1000-8000-00805f9b34fb
Rx Characteristic UUID: 0000ffe2-0000-1000-8000-00805f9b34fb

Patterns: 0x55 0x03 0x01 0x00 0xXX 0x00 (were XX is the pattern 00-0b 00 is off)
Direct control: 0x55 0x04 0x00 0x00 0x00 0xXX (speed 00-0f stops getting faster at some point) only runs for ~1 second

Malaklshook, 2022-08-10:
> BLE Name: AMN NEO Service UUID: 0000ffe3-0000-1000-8000-00805f9b34fb Tx Characteristic UUID: 0000ffe1-0000-1000-8000-00805f9b34fb Rx Characteristic UUID: 0000ffe2-0000-1000-8000-00805f9b34fb [...]

Scarlet-Tsunami, 2022-09-03:
My AMN NEO could connect, but it can‘t turn off，Is it my device problem？
https://s.click.aliexpress.com/e/_ArWoaE

blackspherefollower, 2022-09-03:
> My AMN NEO could connect, but it can‘t turn off，Is it my device problem？ https://s.click.aliexpress.com/e/_ArWoaE

Does it work as expected with the vendor's app?